ottelo9/tasmota-sml-images #1: C3 & C6 - reboot issue with Tasmota 14.4.1

mi-hol, 2025-02-08:
> Hier findest du die aktuell von mir verwendeten Dateien:
> https://github.com/ottelo9/tasmota-sml-images
> 
> Achja, ich habe eben ein Tipp von gemu erhalten:
> 
> also der stack protector Fehler bedeutet dass ein Buffer in der betreffenden subroutine einen Überlauf hat. Warum das nicht bei esp8266 oder normalen esp32 auftritt ist ein Rätsel. in dieser Routine (Replace_Cmd_Vars) gibt es nur einen einzigen Buffer (für Strings char string[SCRIPT_MAX_SBSIZE];).
> versuch mal den größer zu machen in dem du bei >D 48 schreibst ( >D alleine würde den Buffer auf 20 setzen, so geht er auf 48 Zeichen)
> 
> #define SCRIPT_MAXSSIZE 48
> 
> ist der default den du auch noch größer machen kannst, also z.B.
> 
> #define SCRIPT_MAXSSIZE 64
> 
> und >D 64
> 
> Ich habe meine user_config_override.h bereits angepasst und ein C6 Image erstellt. Test läuft. >D 40 habe ich bei mir oben im Script aber berits vorher schon drin gehabt. Ich habe es nun auf 48 erhöht ...
> 
> #define SCRIPT_LARGE_VNBUFF
> habe ich nun auch mit drin.
> 
> Update:
> Hat nix gebracht 

 _Originally posted by @ottelo9 in [#1](https://github.com/ottelo9/tasmota-sml-script/issues/1#issuecomment-2643621259)_

mi-hol, 2025-02-08:
das als Workaround erzeugte Tasmota 14.1.0 funktioniert jetzt seit über einem Tag mit dem C6

Feld | Wert
--- | ---
Datum | 08.02.2025 - 14:07:31
Uptime | 01d 00h 25m
Program Version | 14.1.0 (ottelo.jimdo.de)
Build Date & Time| 2025-02-06T14:29:57
Core/SDK Version | 3_0_0/5.1.4.240529

ottelo9, 2025-02-08:
> das als Workaround erzeugte Tasmota 14.1.0 funktioniert jetzt seit über einem Tag mit dem C6

Kompiliert mit meiner [user_config_override.h](https://github.com/ottelo9/tasmota-sml-images/blob/main/user_config_override.h) und im Script >D 48 verwendet?

Freut mich das es mit deinem Script läuft. Mit meinem Script [2_Tasmota_SML_Script_Chart_PV.txt](https://github.com/ottelo9/tasmota-sml-script/blob/main/2_Tasmota_SML_Script_Chart_PV.txt) und `$<script> setTimeout("location.reload(true);",10000); </script>` in der >W Sektion leider "noch" nicht.

mi-hol, 2025-02-08:
> Kompiliert mit meiner [user_config_override.h](https://github.com/ottelo9/tasmota-sml-images/blob/main/user_config_override.h) und im Script >D 48 verwendet?

Beides nein, Image von deinem GDrive. 
Wegen der Limitation von GDrive sieht man dort leider kein genaues Build Date & Time :(

Bis ich selbst kompilieren kann, dauert wohl noch ein bischen Zeit. Erst musss ich alles richtig konfiguriert habe, deshalb liebe ich GH Actions (GHA) wie zB [FOME firmware build](https://github.com/FOME-Tech/fome-fw/blob/master/.github/workflows/build-firmware.yaml).
Wenn man GHA verwendet, kann man direkt Code Änderungen machen und testen, ohne sich mit dem Aufbau der ganzen dafür notwendigen Umgebung zu belasten!
Ich bin zb kein C Programmierer, konnte aber durch die GHA einfache Korrekturen in FOME erfolgreich einbauen.

ottelo9, 2025-02-08:
> das als Workaround erzeugte Tasmota 14.1.0 funktioniert jetzt seit über einem Tag mit dem C6

Ich dachte du hast noch SML Probleme? 
https://github.com/ottelo9/tasmota-sml-script/issues/1#issuecomment-2642443847
https://github.com/ottelo9/tasmota-sml-script/issues/1#issuecomment-2642649146

mi-hol, 2025-02-08:
> Ich dachte du hast noch SML Probleme?

Du meinst mit unplausiblen Werten? Die sind  bisher nicht wieder aufgetreten

ottelo9, 2025-02-08:
Teste bitte nochmal 14.4.1 für den C6. Gemu hat die Funktion umgebaut, dass kein Stack Overflow mehr passieren kann, das war wohl die Ursache allen Übels. Er weiß aber immer noch nicht, wieso das erst ab Tasmota 14.2.0 und nur auf C3 und C6 passiert.

Hier ist sein Commit
https://github.com/gemu2015/Sonoff-Tasmota/commit/b4ca7dcbf7ddfdf66c385ddfabca695d7abb1739

Ich habe die geänderte `xdrv_10_scripter.ino` bei mir ausgetauscht und ALLE 14.4.1 images neu erstellt.

Hier liegen die neuen Images:
https://github.com/ottelo9/tasmota-sml-images/releases/tag/V14.4.1

mi-hol, 2025-02-09:
> Teste bitte nochmal 14.4.1 für den C6.

Erster Versuch ergab beim Upload Fehler

[image]

mi-hol, 2025-02-09:
leider wieder reboot in Kombination mit unplausiblem Verbrauch!

[image]

ottelo9, 2025-02-10:
Evtl. ist es das Problem hier:
https://github.com/arendst/Tasmota/discussions/22833#discussioncomment-12117817

mi-hol, 2025-02-11:
@ottelo9 wirst Du demnächste neue 14.4 Images mit der oben genannten Fehlerkorrektur anbieten?

ottelo9, 2025-02-11:
Heute Abend. Lade ich die neuen Images hoch. C6 lief die Nacht ohne Reboot durch

mi-hol, 2025-02-11:
updated 2 C6 to https://github.com/ottelo9/tasmota-sml-images/releases/tag/V14.4.1_250211
Lets see for results tomorrow

mi-hol, 2025-02-12:
Bisher stabil!

. | .
-- | --
Program Version | 14.4.1 (ESP32-C6 ottelo.jimdo.de)
Build Date & Time|2025-02-10T17:41:34
Core/SDK Version|3_1_1/5.3.2.250120
Uptime|1T00:16:12

mi-hol, 2025-02-13:
. | .
-- | --
Uptime | 1T13:31:45

mi-hol, 2025-02-15:
läuft zuverlässig, habe jetzt auch automatischen refresh der webseite  einmal je Minute aktiviert

ottelo9, 2025-02-15:
Bei mir läuft es auch einwandfrei auf allen ESP32 Varianten ( C6 C3 )

closed 2025-02-15 by ottelo9