ottelo9/tasmota-sml-images #2: Build Date in Fußzeile der Webseite und Download-Dateien einfügen

mi-hol, 2025-02-15:
zur einfachen und sicheren Identifikation eines Builds wäre es gut das "Build Date" in der Fußzeile der Webseite und den Download-Dateien einzufügen

1. Webseite
Fußzeile heute:
"Tasmota 14.4.1 (ESP32-C6 ottelo.jimdo.de) by Theo Arends (https://bit.ly/tasmota)"

neu:
"Tasmota 14.4.1 (Build:250115)(ESP32-C6 ottelo.jimdo.de) by Tasmota (https://github.com/ottelo9/tasmota-sml-images/releases)"

2. Download-Dateien
Dateiname heute:

[image]

neu:
tasmota32c3_ottelo_250215.zip

ottelo9, 2025-02-23:
Das Build Datum mit an meinen String -DCODE_IMAGE_STR='"ESP32-C3 ottelo.jimdo.de"' in der platformio_tasmota_cenv.ini zu hängen, habe ich nicht gefunden. Anscheinend ist die String Größe auch auf 25 begrenzt. D.h. das Build Datum würde da auch nicht mehr mit rein passen. Ich ändere nicht den Tasmota Code um das da reinzuhacken. Die Info steht ansonsten auch unter "Information -> Build Date & Time".

Das Builddatum in die zip reinzuhängen werde ich aber umsetzen, da sinnvoll.

Falls jemand eine Lösung findet, baue ich sie gerne ein.

closed 2025-02-23 by ottelo9

mi-hol, 2025-02-23:
Danke, das Builddatum in die zip reinzuhängen bringt schon viel!

Insgesamt bisher 300 downloads ist schon ein sehr gutes Ergebnis!

[image]