ottelo9/tasmota-sml-images #4: please add "USE_SCRIPT_FATFS_EXT" to allow saving of data to file

mi-hol, 2025-02-19:
wegen auf 1536 Byte begrenztem "permanenten Variablen" hat gemu2015 das s[peichern von grossen Arrays in Dateien als Lösung vorgeschlagen](https://github.com/ottelo9/tasmota-sml-script/issues/6#issuecomment-2667688605).
Das funktioniert aber nur wenn das image mit USE_SCRIPT_FATFS_EXT erstellt wurde.

Bitte nimm  USE_SCRIPT_FATFS_EXT in die build config auf.

mi-hol, 2025-02-19:
@ottelo9 Vielen herzlichen Dank!
Jetzt funktioniert das sichern von SML Daten für ein Jahr durch schreiben in eine Datei!

mi-hol, 2025-02-19:
1000 Zeile code mit Kommentaren, geben 9697 Byte script.txt, die gespeicherten Daten werden ca 8 kB brauchen

Flash Size	4096 KB
Program Flash Size	4096 KB
Program Size	1108 KB
Free Program Space	1771 KB
Free Memory	279.6 KB (frag. 11%)
Partition safeboot	832 KB (used 96%)
Partition app0*	2880 KB (used 38%)
Partition fs	320 KB

mi-hol, 2025-02-21:
nach 3 Tage Laufzeit
<html>
<body>

Free Memory | 278.5 KB (frag. 8%)

</body>
</html>

ottelo9, 2025-02-23:
Im Image vom 23.02.2025 enthalten.

closed 2025-02-23 by ottelo9