ottelo9/tasmota-sml-images #5: Build Tasmota firmware 14.5.0 Ruth  ?

mi-hol, 2025-02-23:
Ist ein Image für die seit 18 Feb verfügbare Tasmota firmware 14.5.0 Ruth  geplant?

ottelo9, 2025-02-24:
Erstmal nicht. Brauchst du unbedingt das neue Update?

mi-hol, 2025-02-24:
Ich wollte nur sehen, ob es neue Probleme mit der Version gibt.
Generell, zu meinem Verständnis: Wann planst Du  neue Versionen zu publizieren (z.B n Wochen nach Freigabe,..)?

ottelo9, 2025-02-25:
Da es alles in meiner Freizeit geschieht und ich eigentlich wenig Zeit für diese Art von Hobbis habe, plane ich nichts. 

Aber gelegentlich schaue ich nach, ob’s ne neue Version gibt, wo es sich lohnt Images zu erstellen. Oder jemand schreibt mich über mein Blog an, dass er unbedingt die neue Version benötigt.

 Das Auffinden des Rebootsbugs hat mich viel zu viel Zeit gekostet, die ich eigentlich nicht hatte.

closed 2025-02-25 by ottelo9

mi-hol, 2025-02-25:
Aber warum schliesst Du es dann gleich? Lass dir doch helfen! 
Man signalisiert das durch das Label "help wanted".

mi-hol, 2025-03-07:
@ottelo9 magst Du dir da nicht helfen lassen? "reopen issue" kannst nur Du machen!