ottelo9/tasmota-sml-images #6: track pmem scripting patch  until merged into upstream (commit 2d6fb46067677cf9ac4b62078ea799823b84fdc4)

mi-hol, 2025-02-23:
Damit nicht 2 Zweige von Tasmota scripting entstehen, sollten wir den mergen von commit 2d6fb46067677cf9ac4b62078ea799823b84fdc4  beobachten.
Bisher scheint kein PR dafür zu existieren

ottelo9, 2025-02-24:
Aktuell verwendete Scripter File im letzten Build: https://github.com/ottelo9/tasmota-sml-images/blob/main/xdrv_10_scripter.ino

mi-hol, 2025-04-15:
seems merged via https://github.com/arendst/Tasmota/commit/2e35a2ab936d873f9c235232e77785f6e29c95f6 into Tasmota's  Development branch

mi-hol, 2025-04-23:
Aus meiner Sicht entfällt durch die Tasmota Version 14.6.0 die Notwenigkeit für
"TasmotaProjekt/tasmota/tasmota_xdrv_driver/xdrv_10_scripter.ino <- (optional) die aktuellste Scripter Source aus der [gemu2015 Repo](https://github.com/gemu2015/Sonoff-Tasmota/blob/universal/tasmota/tasmota_xdrv_driver/xdrv_10_scripter.ino)"

@ottelo9 Oder möchtest Du tatsächlich 2 Zweige von Tasmota scripting haben?

ottelo9, 2025-04-27:
Wie ich [hier](https://github.com/ottelo9/tasmota-sml-images/releases/tag/V14.6.0_250423) geschrieben habe, hat gemu mir DeepSleep Funktionen eingebaut. Das wollte ich in diesem Release mit drin haben. Deshalb habe ich die Datei hier mit abgelegt. 

Wenn das beim nächsten Release nicht mehr nötig ist, lasse ich sie natürlich weg.

mi-hol, 2025-04-27:
Als ich den Kommentar geschrieben hatte, war mir das noch nicht bekannt.
Danke für die Klarstellung.