ottelo9/tasmota-sml-images #10: >J funktioniert nicht --> Keine zusätzlichen Werte im Status bei MQTT

dirkheitzmann, 2025-03-21:
Erstmal, tolle Scripts. Danke dafür.

Bei meinem alten Script hatte ich auch eine ">J" Section.
Damit kann man zusätzliche Werte nach MQTT schicken:

```
>J[~](https://tasmota.github.io/docs/Scripting-Language/#j)
The lines in this section are published via MQTT in a JSON payload on [TelePeriod](https://tasmota.github.io/docs/Commands/#teleperiod). Requires compiling with #define USE_SCRIPT_JSON_EXPORT.
```

Vielleicht könnte man zukünftig das "#define USE_SCRIPT_JSON_EXPORT" zusätzlich einbauen ?

ottelo9, 2025-03-21:
Werde ich im nächsten Release mit einbauen oder brauchst du es jetzt sofort?

dirkheitzmann, 2025-03-21:
Keine Panik, hab mir das aktuelle 14.5 mit den Libs von Dir, der Erweiterung und was sonst noch gebraucht wurde schon selbst gebaut.

ottelo9, 2025-04-23:
https://github.com/ottelo9/tasmota-sml-images/releases/tag/V14.6.0_250423