ottelo9/tasmota-sml-images #14: Unterstützung ESP32-S3-POE-ETH mit POE

PLCHome, 2025-04-22:
Hallo, ich habe diesen ESP32-S3-POE-ETH gefunden. 

https://www.waveshare.com/esp32-s3-eth.htm?sku=28771
https://de.aliexpress.com/item/1005008524380174.html

[image]

Da ich in diesem Keller kein WLAN habe, hatte ich gedacht ETH und POE wäre eine gute Idee.

A) Der ESP läuft gut über POE.
B) Dieses Tasmota läuft auf ihm gut über WLAN.

Aber...  Ich brauche ein Image, das beides kann..

Daher die Frage ist es dir möglich, auch ein Image für den ESP32-S3 mit ETH wie für den ESP32 zur Verfügung zu stellen?

PLCHome, 2025-04-22:
Danke, du bist echt der Beste

ottelo9, 2025-04-22:
Kannst du mal bitte testen, ob's hiermit klappt?
[tasmota32s3_ottelo_250422.zip](https://github.com/user-attachments/files/19855746/tasmota32s3_ottelo_250422.zip)

PLCHome, 2025-04-22:
ETH sieht gut aus.

[image]

SML habe ich jetzt nicht getestet. Das habe ich an dem ESP noch nicht dran.

PLCHome, 2025-04-23:
War gestern etwas spät. 
Hier noch mal die ganze Geschichte:

Zu Anfang wollte der ESP nicht.
Ich hatte das BIN mit OTA über WLAN eingespielt, danach war er nicht mehr erreichbar. 
Allerdings war das Netzwerkkabel nicht richtig eingesteckt :-(

Danach hatte ich dann die factory.bin über USB mit dem Webflasher eingespielt. Das endetet in eine Boot Endlosschleife.
Frustriert hab ich dann Tasmota32-S3 (OPI Flash/OPI PSRAM) vom Webflasher eingespielt und noch mal das BIN mit OTA über WLAN eingespielt, danach lief er.... 

Alles komisch aber jetzt scheint er zu laufen.

Du hast mir mit dem Image echt viel Arbeit erspart. 
Heute Abend werde ich mal den Lesekopf löten und testen. 

Die Schaltung ist für meinen Schwager, da ist alles nicht so einfach, weil alles im Keller verteilt ist und das WLAN echt mies ist. 

Bei mir habe ich das Glück, den Server in der Nähe zu haben. Da habe ich einfach die Infrarotschnittstelle unter dem Zähler über eine Westernkupplung auf einen RS232 zu USB Adapter, der auch 5V rausgibt gelegt und lese das einfach über den ioBroker über USB aus. Die Schnittstelle hatte ich damals in den Sockel im Schrank eingebaut. Solange das EVU die nicht benutzt, habe ich mir gedacht, bezahlt habe ich die ja...

ottelo9, 2025-04-23:
https://github.com/ottelo9/tasmota-sml-images/releases/tag/V14.6.0_250423

closed 2025-04-23 by ottelo9