ottelo9/tasmota-sml-images #16: Current power of a meter, delay about 3 seconds

maroviher, 2025-04-29:
@ottelo9, erstmal Danke für dein Projekt. Was ich erreichen will ist, dass die Momentanleistung des Zählers mit der minimal möglichen Verzögerung per MQTT oder sonst irgendwie, an ein Host in meinem LAN geschickt wird. Ich habe es so probiert:
```
>M 1
+1,3,s,20,9600,E320
1,77070100100700ff@1,Current power,W,Power_in,19
```
So schickt Tasmota die MQTT Messages 1 Mal in der Sekunde aber die Daten des Momentanverbrauchs sind ca. 3 Sekunden veraltet gegenüber der Anzeige am Zähler. Was mir auch reichen würde ist, wenn der ESP32 den Datensatz über die Infrarotschnittstelle empfangen hat, dass er diese Daten sofort per MQTT weiterleitet, sodass es nicht zu Verzögerung im ESP32 kommt. Hättest Du eventuell einen Tipp? Danke!

maroviher, 2025-04-29:
Also Median Filter scheint schuld daran zu sein. Hab jetzt 
+1,3,s,20,9600,E320
auf
+1,3,s,0,9600,E320
geändert und jetzt ist es flotter

ottelo9, 2025-04-29:
Hi. Gerne solche Fragen in mein Blog stellen, unten im [Gästebuch](https://ottelo.jimdofree.com/stromz%C3%A4hler-auslesen-tasmota/#Kommentare). Das hier ist eher für Probleme gedacht.